piotrC4/mqtt-ir-transceiver #2: INO file

bachoo786, 2017-01-10:
Hi

Thanks for this project

I was trying to find the INO file/sketch to upload it to my Arduino ESP8266 board. Can you please upload that?

Thanks.

enc-X, 2017-01-10:
Folders structure and filenames are compatibile with Platformio not for Arduino IDE. You can try use main.cpp from src folder as main file and add other files to Arduio project, also add required libraries and try compile - but I never test it with Arduino.

bachoo786, 2017-01-11:
Hi
Thanks for your reply.
I wanted to find out I have a node mcu esp8266 board and was wondering what platform can I use to install the libraries other than arduino? Are there any softwares to install the libraries? 
Also where do I change the WiFi configuration and mqtt settings?in which file?

Thanks

enc-X, 2017-01-12:
If you use platformio all required libraries will install during code compilation - they are defined in platformio.ini file. If you use Arduino IDE - you have manually install libraries listed in README.
You don't have to change WiFi configuration in any file - WiFiManager will put device in configuration mode. You only have connect to it from you PC or mobile (device will act as AP) and fill in configuration form.

bachoo786, 2017-01-12:
ok so I have installed Platformio and have installed all the libraries (IRremoteESP8266, ArduinoJson, PubSubClient, WiFiManager), copied the src folder and the the platformio,ini file. 

This is my first time using Platformio. 

My question is do I need to copy your code that you posted on here: https://community.openhab.org/t/mqtt-ir-transmiter-receiver/9433

or is this code that you posted only for Arduino devices? 

If I do have to copy the code, can you please tell me how and on what file shall I write it to?

Many thanks.

enc-X, 2017-01-12:
It is for ESP, only framework is Arduino compatible.  I forged write that I'm using atom editor with platformio plugin. So you have  to download whole folder and import it as platformio project - and from platformio menu you can compile and upload image

bachoo786, 2017-01-12:
ok I am using and Atom as well.

this code: https://community.openhab.org/t/mqtt-ir-transmiter-receiver/9433
I dont need it, is that correct?

after uploading the image from platformio I have to go to 192.168.4.1 and connect my wifi and mqtt, is that correct?

Thanks.

enc-X, 2017-01-12:
The source code from openhab site it is old one - yes, you don't need it.
And yes - after connection to 192.168.4.1 you will be able to configure wifi and mqtt.

bachoo786, 2017-01-12:
Hi

Thanks alot

May I ask what is MQTT prefix? is it the port at which MQTT is connected to?

enc-X, 2017-01-12:
It is prefix of MQTT topic used for communication with module.

bachoo786, 2017-01-12:
Oh OK do I have to put anything in the prefix? As I have uploaded the coded and connected it to the WiFi but not sure if it's connected to my mosquitto as I connected my IR Receiver to my nodemcu v2.0 on pin D7 and subscribed to topic: _mqtt_prefix_/receiver/RC_5/12

bachoo786, 2017-01-12:
I try and turn my Samsung TV on but I receive no code from IR Receiver on mqtt.

Do you know what could be the problem?

bachoo786, 2017-01-12:
Can you please tell me how do I know if I am connected to my mqtt server? 

And also can you give a very simple example of how to receive and to send IR codes ?

Please help !

Thank you

ronaldotz90, 2017-01-13:
Hello there @enc-X 

I have the same problem as @bachoo786. 

I have uploaded the code to my ESP8266 and I have connected it to WiFi but my problem is I am not sure if its connected to my mosquito server which runs on my rpi. Is the mqtt port : 1883?

Can you please show me how to subscribe to a topic and receive IR code correctly?

Thank you so much.

enc-X, 2017-01-13:
I don't how deeply you know MQTT concept - here: http://www.steves-internet-guide.com/mqtt/ you can find very nice introduction to this protocol.
In IR Transceiver when you setup prefix as for example: 'mydevices/irtransmiter', proper username and password and connect to MQTT server you can:

Send IR code:
`mosquitto_pub -u <server_user> -P <server_password> -t "mydevices/irtransmiter/sender/sendGC" -m "38000,1,69,341,170,21,21,21,21,21,64,21,21,21,21,21,21,21,21,21,21,21,64,21,64,21,21,21,64,21,64,21,64,21,64,21,64,21,64,21,21,21,21,21,21,21,64,21,21,21,64,21,64,21,21,21,64,21,64,21,64,21,21,21,64,21,21,21,21,21,1517,341,85,21,3655"`

You can subscribe device topic to see what IR trasmission are received:
`mosquitto_sub -u <server_user> -P <server_password> -v -t "mydevices/irtransmiter/#"`

If you use Wemos-D1 you can enable debug and see what's going on via serial port.

bachoo786, 2017-01-14:
Hi @enc-X 

Thank you so much.

I receive my samsung tv code when I turn on my TV using this:
 
mosquitto_sub -u <server_user> -P <server_password> -v -t "mydevices/irtransmiter/#” 

The code is: 

mydevices/irtransmitter/receiver/SAMSUNG/32 -522174273

Now I want to ask you can I use this code that to you gave to turn on the TV?

This one:

mosquitto_pub -u <server_user> -P <server_password> -t "mydevices/irtransmiter/sender/sendGC" -m “38000,1,69,341,170,21,21,21,21,21,64,21,21,21,21,21,21,21,21,21,21,21,64,21,64,21,21,21,64,21,64,21,64,21,64,21,64,21,64,21,21,21,21,21,21,21,64,21,21,21,64,21,64,21,21,21,64,21,64,21,64,21,21,21,64,21,21,21,21,21,1517,341,85,21,3655"

or does it have to be : 

mosquitto_pub -u pi -P raspberry -t "mydevices/irtransmitter/sender/RC_5/12" -m "-522174273”

I have read the article you sent but it does not talk about MQTT topics.

Can you please help?

Thanks

thebgrian, 2017-01-14:
bachoo786, try the following:

mosquitto_pub -u <server_user> -P <server_password> -t "mydevices/irtransmiter/sender/SAMSUNG/32" -m "-522174273"

The codes I receive and send don't have "-" in front, but whatever you receive from your remote is what you need to send.

In some cases you won't see any codes recognized  from the receiver and in this case you can use the Global Cache codes that look like this (this is the one I use for a DirecTV receiver). You can register on the https://irdb.globalcache.com/ web site, but that didn't work for me for some reason. I jus googled for the codes I needed - they should be in a simple text file.

topic: mydevices/irtransmiter/sender/sendGC
message: 39000,1,1,231,46,46,46,23,23,23,23,23,23,46,46,23,46,46,23,46,23,23,1143,115,46,46,46,23,23,23,23,23,23,46,46,23,46,46,23,46,23,23,1143.

If you use Chrome - there is a great extension - MQTT Lens that you can use for testing the commands

bachoo786, 2017-01-14:
Hi @thebgrian 

I have tried it all and nothing seems to work.

Can I ask you the code that I received : "-522174273” what format is it on? is it for NEC? or Samsung or what?

Nothing seems to work. 

When I use a normal LED and use the code: mosquitto_pub -u <server_user> -P <server_password> -t "mydevices/irtransmiter/sender/SAMSUNG/32" -m "-522174273” 

The LED flashes but when I change it to a IR LED with the same code, it doesnt turn the TV on/off.

I have even googled the GC codes for my Samsung TV and they dont seem to work. I guess its the IR library that its not allowing the codes to work.

thebgrian, 2017-01-14:
If the codes are correct and recognized by the transmitter they will be posted to the mydevices/irtransmitter/ topic.
Just subscribe for mydevices/irtransmitter/# topic in a separate session and watch for the confirmation. If you see them, then check the direction of your IR diodes. I had the transistor wrongly wired the first time, but if you can see normal LEDs blinking you should be good.

enc-X, 2017-01-14:
You answerd all questions - thanks ... I just want to add that I didn't test Samsung compatibility (I have LG NEC and RC5 compatible devices).  Value lower than 0 may mean that it is out of range and if you use it for sending IR you may will have no result. The best option to send IR codes to device is usage of Global Cache database like @thebgrian wrote every device I tested with GC worked well - I have even more features than with standard codes (for example switching to specific input).

bachoo786, 2017-01-14:
Hi 

Thank you both for yours answers.

I have a question @enc-X 

Are you saying this system will work with “Philips RC5 Protocol” ??

Thanks.

thebgrian, 2017-01-14:
Thank you for the great project enc-X. MQTT is so universal that you can use this tranceiver with many other systems.

bachoo786, 2017-01-14:
When you say RC5 do you mean Philips RC5 Protocol ?

enc-X, 2017-01-14:
Yes - I tested with Philips amplifier.

bachoo786, 2017-01-17:
Hi @enc-X 

Thanks for your reply.

I have emulated the circuit you have shown in the main page and to my surprise the led doesnt work. 

I have checked my circuit 10x and everything is fine. If I connect the IR LED to D5 and GND it works perfectly but when I connect it to a transistor (2N2222 NPN), 2x 1N4148 Diodes, 3.3kohm and 100ohm resistors it doesnt work.

Can you please check the circuit or tell me what am I doing wrong please? 

Many thanks.

ronaldotz90, 2017-01-18:
@bachoo786  have you checked your transistor connection, you could have mixed the pins up?

enc-X, 2017-01-18:
@bachoo786 - like @ronaldotz90 wrote - check transistor wiring. I'm using C33740 so 2n2222 shoud work.

bachoo786, 2017-01-19:
Hi @enc-X 

I had mixed the transistor pins up its noway working fine. What range are you getting with your circuit? 

Also I got the GC codes for my Samsung TV but surprisingly it doesn't work. What I mean is that the topic in general seems unresponsive as I connected a normal LED and it didn't work.

Do you know what could be the problem?

Many thanks again.

enc-X, 2017-01-19:
@bachoo786 please provide example of GC code which doesn't work - I'll test.

bachoo786, 2017-01-19:
Here is the code: 

sendir,1:1,1,38000,1,1,172,172,22,64,22,64,22,64,22,21,22,21,22,21,22,21,22,21,22,64,22,64,22,64,22,21,22,21,22,21,22,21,22,21,22,64,22,21,22,21,22,64,22,64,22,21,22,21,22,64,22,21,22,64,22,64,22,21,22,21,22,64,22,64,22,21,22,1820

Also what range are you getting with your circuit?

enc-X, 2017-01-19:
I send by using command:
`mosquitto_pub -u <user> -P <pass>  -t "esp8266/999/sender/sendGC" -m "38000,1,1,172,172,22,64,22,64,22,64,22,21,22,21,22,21,22,21,22,21,22,64,22,64,22,64,22,21,22,21,22,21,22,21,22,21,22,64,22,21,22,21,22,64,22,64,22,21,22,21,22,64,22,21,22,64,22,64,22,21,22,21,22,64,22,64,22,21,22,1820"`
and I see blinking IR LED (via mobile phone camera)

enc-X, 2017-01-19:
I also tested range - about 7 meters with my TV

travplan, 2017-01-19:
I've got this thing working for the most part. Im using MQTTfx to publish messages to the Adafruit HUZZAH with an esp12 (Im also using openhab2, but for quick testing a quick publish from MQTTfx is easier). I see in the serial monitor in platformio that the messages are being recieved, but nothing seems to be outputting. I'm using pin "0" for the IR output (configured in globals.h under the "production device - ESP01") just because it has an led on it, in the hopes of getting a flash i can see. I also have an oscilloscope hooked up to pin "0". No matter what i send, i see nothing on either the scope or the led. 
Here is what I'm seeing in the serial monitor... i copied one of your publish examples exactly, just to get the greatest chance of something working.
*IR: ======= NEW MESSAGE ======
*IR: Topic: "esp8266/02/sender/RC5/12"
*IR: Message: "56"
*IR: Length: 2

*IR: Prefix: esp8266/02
*IR: Extracted suffix:"/sender/RC5/12"
*IR: Payload String: "56"
*IR: TypStr=RC5
*IR: irBitrStr=12
*IR: irPanasAddrStr=
*IR: Send RC5:56 (bits: 12)

What am i doing wrong?
Thanks

bachoo786, 2017-01-19:
@enc-X do I have to use "esp8266/999/sender/sendGC” ??

enc-X, 2017-01-19:
@bachoo786 - it depends on configured prefix - my is 'esp8266/999'

enc-X, 2017-01-19:
@travplan - if you changed under 'production device - ESP01' you changed if for version without debug, so I'm guessing that it's not working. If you using debug, change under 'dev device (wemos)'

bachoo786, 2017-01-19:
@enc-X I havent changed anything. how do you configure the prefix?

enc-X, 2017-01-19:
@bachoo786 -  during first configuration of device. Reset to factory defaults (by pressing button during boot) and do initial configure via webpage.

bachoo786, 2017-01-19:
@enc-X ok my prefix during the set up is : mydevices/irtransmitter

enc-X, 2017-01-19:
@bachoo786 - so your topic should be: "mydevices/irtransmitter/sender/sendGC"

travplan, 2017-01-19:
@enc-X thanks, so i assume the "X" beside DEBUG denotes the wemos settings for my HUZZAH. 
Great work on this by the way.
That was great advice, I now have it sending messages and being picked up by another module running a IRdump sketch!! This is really exciting.

One thing I am struggling with is the Raw codes, and Global Cache. what exactly should my string look like to send these? And since im able to grqab the raw codes with my IRdump module, can i just send the raw codes using the sendGC command?

travplan, 2017-01-19:
@enc-X 
When i send the global cache codes, only the first 20-25 elements send. If I send more than that, the serial monitor doesnt actually show anything as received, but i can confirm its being sent watching MQTTfx. Is there some sort of limit?
Here is what im using in my openhab items:
Switch Yam_VolUp "Volume" <receiver> (TV) {mqtt=">[mosquitto:esp8266/02/sender/sendGC/:command:*:38000,1,1,172,172,22,64,22,64,22,64,22,21,22,21,22,21,22,21,22,21,22,64,22,64,22,64,22,21,22,21,22,21,22,21,22,21,22,64,22,21,22,21,22,64,22,64,22,21,22,21,22,64,22,21,22,64,22,64,22,21,22,21,22,64,22,64,22,21,22,1820]"}
Thanks in advance

bachoo786, 2017-01-19:
@enc-X thanks will try it.

Can I ask you how have you powered your wemos d1 in your setup? Is it via the 5v usb or have you powered it via 5v Vin pin?

Thanks

enc-X, 2017-01-20:
@travplan - I have to check sending of raw messages, I have not time to test it well, because I focused on GC. I'll check raw mode little bit more during weekend..
I tested with my openhab you item and in transceiver log everyting looks ok - number of items is ok:
`*IR: ======= NEW MESSAGE ======
*IR: Topic: "esp8266/999/sender/sendGC"
*IR: Message: "38000,1,1,172,172,22,64,22,64,22,64,22,21,22,21,22,21,22,21,22,21,22,64,22,64,22,64,22,21,22,21,22,21,22,21,22,21,22,64,22,21,22,21,22,64,22,64,22,21,22,21,22,64,22,21,22,64,22,64,22,21,22,21,22,64,22,64,22,21,22,1820"
*IR: Length: 217
*IR: Prefix: esp8266/999
*IR: Extracted suffix:"/sender/sendGC"
*IR: Payload String: "38000,1,1,172,172,22,64,22,64,22,64,22,21,22,21,22,21,22,21,22,21,22,64,22,64,22,64,22,21,22,21,22,21,22,21,22,21,22,64,22,21,22,21,22,64,22,64,22,21,22,21,22,64,22,21,22,64,22,64,22,21,22,21,22,64,22,64,22,21,22,1820"
*IR: TypStr=sendGC
*IR: irBitrStr=
*IR: irPanasAddrStr=
*IR: Start parsing message
*IR: Send GC
*IR: Elements to send: 71
*IR: GC send done.`

Maybe it happens bacause of default MQTT_MAX_PACKET_SIZE. Do you platformio? Please check `build_flags = -DMQTT_MAX_PACKET_SIZE=1500` in platformio.ini file.

enc-X, 2017-01-20:
@bachoo786 - I'm using both methods, both are ok.

travplan, 2017-01-20:
@enc-X  - Thanks for the response. 
I checked the MQTT_MAX_PACKET_SIZE in platformio. It was set to 1500. I changed it to 5000 and re-flashed the huzzah as a test. This did not seem to help, the MQTT message is still not seen in the serial monitor, but is sent via MQTT. I'm not really sure what to look for next. Maybe it's a limitation of the esp12 on the huzzah, but i can't imagine.
Are there any other settings or parameters i may have missed?

travplan, 2017-01-20:
@enc-X  - I've narrowed it down.
It is not a limit on how many elements can be sent, it is a limit of "IR: Length" it will only parse out a maximum length of 100 characters. Anything above that is not shown in the serial monitor. Where is this limit set? Thank you

travplan, 2017-01-20:
@enc-X  - Leave a guy to their own devices and soimetimes it works out! 
Success!! It seems that the pubsubclient.h MQTT_MAX_PACKET_SIZE was the limiting factor. It was set to 128 as default. I set it, and the  MQTT_MAX_PACKET_SIZE  found in platformio.ini to 1500, and all is right in the world. 
Thanks again, and it looks like im good to go for the sendGC commands. I'm sure I'll be back at some point with more questions.

enc-X, 2017-01-20:
Great to hear. Setting in platformio.ini shuld overwrite pubsubclient limit because it appear as compilation level - did you do this at this level?

ronaldotz90, 2017-01-20:
@travplan were you not able to send GC commands? because I am having the same problem. what changes did you make to sendGC codes?

travplan, 2017-01-20:
@enc-X - I looked in platformio,ini in the atom interface to start with, but it was already set at 1500 as default, so im not sure why it didnt overwrite it. Either way, it seems to be working now, and that's really all I care about haha.

@ronaldotz90 - I assume you are successful in sending other commands? such as "your_topic_prefix"/RC_5/12" with some sort of message (example: "1234"). If so, try sending a short "your_topic_prefix"/sendGC with a short message like "1,2,3,4". If that is received and works, you may have the same issue i had. If that worked try sending a message with exactly 100 characters. if that works, then try 101. if that doesnt then im sure it's the MQTT_MAX_PACKET_SIZE variable set to "128" (change to "1500") in "pubsubclient.h" in the folder you extracted "mqtt-ir-tranceiver-master.zip" to. It's in the ".pliolibdeps" folder (save the file after changing). I would also make sure the MQTT_MAX_PACKET_SIZE in platformio.ini is at "1500", then rebuild and upload.

Hope that helps.

ronaldotz90, 2017-01-20:
@travplan does the comma count as a character as well or is it just the number/s? i.e. 2,33.....

travplan, 2017-01-20:
@ronaldotz90 - everything that takes up a space is seen as a character. So in your example "2,33" is 4 distinct characters. 
Are you able to open the serial monitor and view the serialprint output of your esp module?

bachoo786, 2017-01-21:
thank you soo much everything works well :)

travplan, 2017-01-21:
@enc-X  - I've been struggling for many hours trying to get the raw code portion working. I have a satelitte receiver box that doesnt seem to respond to any global cache protocol i can find. I do have a remote, so I've been recording the raw codes and trying to send them to the receiver box, it never responds to anything. Just wondering if you had any advice. Thanks

enc-X, 2017-01-22:
I just found that there is problem with RAW recordings - I have to test it a little bit more. In the meantime I changed raw mode support. You can catch RAW sequenc by setting on raw-mode (topic: _mqtt_prefix_/sender/rawMode value 1 - respose will be published in topic: _mqtt_prefix_/receiver/raw) and then send raw code by publishing this value in topic _mqtt_prefix_/sender/sendRAW	 for example:

> mosquitto_pub -u <user> -P <pass>  -t "esp8266/999/sender/rawMode" -m "1"
> mosquitto_pub -u <user> -P <pass>  -t "esp8266/999/sender/sendRAW" -m "9000,4550,550,600,600,600,550,1750,550,600,600,600,600,600,550,600,550,650,550,1750,550,1750,600,550,550,1750,550,1750,600,1700,550,1750,550,1750,550,600,600,600,550,650,550,1750,600,550,550,600,550,650,550,600,550,1750,600,1700,550,1750,550,600,600,1700,550,1750,550,1700,550,1750,550"

I've just tested this raw codes with my TV - works well.

enc-X, 2017-01-22:
I'm closing this 'multi' issue - please open separate request for each issue. It will be easier to manage them.

closed 2017-01-22 by enc-X