piotrC4/mqtt-ir-transceiver #3: Mqtt IR Transceiver Power issues

bachoo786, 2017-01-23:
Hi

I have managed to get everything working after a lot of hurdles. I have noticed one issue which I think can be improved if possible.

Whenever I restart my raspberry pi which has the mosquitto running on it. The nodemcu does not seem to respond or connect to the mosquitto after it has restarted until I switch the nodemcu off and on it again. This seems to be a real issue as the pi can restart on its on after an update or at times requires rebooting due to technical issues.

Is there anyway possible that you could modify in your code to make the nodemcu/esp8266 connect straight to the mosquitto after it reboots?  

NB: nodemcu looses the Mqtt connection not WiFi.

Thanks.

travplan, 2017-01-24:
@bachoo786 - Might be a stupid question as I have no experience with rasp pi... but are you using static ip settings on your **pi?** Also make sure your mosquitto.conf file is set to your static ip, not "localhost" ( I had issues with that before). The only time I've had any issues with MQTT reconnections was with conflicting ip settings. Otherwise, it's all been rock solid.

bachoo786, 2017-01-24:
@travplan - I am using a static IP on my pi as its connected via wifi. I will have to look into the mosquitto.conf. Are you saying your client running mqtt-ir-transceiver is having no re connection problems after your MQTT broker has rebooted? i.e. you dont need to reboot your client running mqtt-ir-transceiver?

travplan, 2017-01-24:
@bachoo786 - I will double check for you when I get home later today, but I'm sure I haven't had that problem. Line 134 in you mosquitto.conf is where you set your broker ip to match you rasp pi's static ip.

enc-X, 2017-01-24:
Yes - the situation may occure. It is undocumented yes mode - "nonMQTT". I will upload new code shortly which in case on MQTT disconnect every 10 sec try to reconnect.

bachoo786, 2017-01-24:
@travplan thanks for that.
@enc-X I am confused by your comment !

enc-X, 2017-01-24:
@bachoo786 new version is on place. I didn't test it very carefully yet, give me some time.
This feature comes from my need to periodically send some codes even device is offline.

bachoo786, 2017-01-24:
@enc-X so you are saying MQTT would work even if the broker is offline? Thanks for the update !

enc-X, 2017-01-24:
Device will try to reconnect to broker every 60 seconds if it is disconnected. Old version tried 4 times in 15-20 seconds - so if you disabled broker for more then 15-20sec. device was not able to connect till reboot.

bachoo786, 2017-01-24:
@enc-X that is perfect. Can you please remove the command of rebooting the client completely.
Can't wait for the update.

enc-X, 2017-01-24:
I think  that we have here small misunderstood.  In previous post I described nature of problem which was present only in old version of software. Current version should work - and client will try reconnect automatically.

bachoo786, 2017-01-24:
@enc-X So are you saying the current version will try to connect every 60 seconds? Because when my broker reboots the mqtt client does not connect until I reboot the client.

enc-X, 2017-01-24:
Strange ... I've just tested with 2 brokers and in debug I have something like this (entering non MQTT and returning to MQTT mode):

> *IR: MQTT connect failed, rc=-2 try again in 5 seconds
> *IR: MQTT user:xxxxxxxx
> *IR: MQTT pass: ********
> *IR: MQTT connect failed, rc=-2 try again in 5 seconds
> *IR: Entering non MQTT mode
> *IR: Not connected to MQTT....
> *IR: Connecting to: mqtt.lan as IRGW-xxxxxxxxxxx
> *IR: MQTT user:nodemcu
> *IR: MQTT pass: ********
> *IR: Connected to MQTT broker
> *IR: Topic is: esp8266/999/sender/#
> *IR: Successfully subscribed
> *IR: Entering MQTT mode

In my test I connected esp to broker, then stop broker, esp entered into non MQTT mode, I started broker, after about 60 sec esp was connected to broker again.

bachoo786, 2017-01-24:
@enc-X surprisingly I reboot my broker but the esp doesn't connect unless I reboot it

enc-X, 2017-01-24:
Please check if you have correct version - newest one publish software version number during boot:

> esp8266/999/info/version 0.01

bachoo786, 2017-01-24:
@enc-X how do I check that?

enc-X, 2017-01-24:
You should subscribe all messages published by device.

bachoo786, 2017-01-24:
@enc-X it's not working just tried it. I have to reboot the esp8266 and then it works.

@travplan did u check if you are having the same issue as me?

bachoo786, 2017-01-24:
@travplan line 134 in the mosquitto.conf is on the image attached.
[image]

bachoo786, 2017-01-24:
@enc-X when is the new version out and how is it different from the current version in terms of reconnecting to the broker? 
Thanks

enc-X, 2017-01-25:
It's already online. I added it in commit b009048

travplan, 2017-01-25:
@bachoo786 - sorry I didnt get a chance to take a look last night, I'm in the middle of some major home renovations, which often takes precedence (especially in the wife's eyes!).
In your image attached, it would be the commented line 132 add a line below it and leave the commented line). See my configuration below.
I'm not saying this is your problem, but I had similar problems with a different device without setting this.

`# =================================================================
# Default listener
# =================================================================

# IP address/hostname to bind the default listener to. If not
# given, the default listener will not be bound to a specific 
# address and so will be accessible to all network interfaces.
# bind_address ip-address/host name

bind_address **my.computer.ip.address**

# Port to use for the default listener.

port 1883

bachoo786, 2017-01-25:
@enc-X I have just updated to the latest software and can confirm everything is working fine. I rebooted my raspberry pi (mosquitto broker) and my esp8266 (client) did connect straight away after a few seconds i.e. I did not have to reboot my esp8266 (client). 

Again Thank you ever so much.

@travplan I did not need to change the mosquitto.conf as the software update of “Mqtt IR Transceiver” done the trick but thanks anyway.

closed 2017-01-25 by enc-X

bachoo786, 2017-01-25:
@enc-X Can I just mention after I reboot my broker and the client connects automatically after a few seconds, why is it that an IR signal is sent automatically without me sending any IR signal?

enc-X, 2017-01-25:
The initial code was prepered for sending IR codes stored on SPIFFS periodically . And you saw this feature. I pushed version with disabled auto sending.
PS - please do not comment under old issues - please open new one.