piotrC4/mqtt-ir-transceiver #4: MQTT-IR-Transceiver Connection

bachoo786, 2017-01-25:
Hi @enc-X 

From my previous post, I mentioned that an IR signal is sent after the client reconnects to the broker and apologies for my insufficient knowledge but is there anyway that you could see into or change the code that would disable sending an IR signal after the client reconnects to the broker?

Thanks.

enc-X, 2017-01-25:
It's already done - you should have version 0.03

bachoo786, 2017-01-25:
@enc-X did you just do it now as we speak? like 5 mins ago?

enc-X, 2017-01-25:
Yes - current one is 0.03

bachoo786, 2017-01-25:
perfect I will try it now and let you know the outcome. Thanks again !

bachoo786, 2017-01-25:
@enc-X I cannot build the software it gives me this error:
[image]

enc-X, 2017-01-25:
Try now - it was letters size of filename problem.

bachoo786, 2017-01-25:
tried it but have the same issue as before: the IR signals are sent right after the client connects to the broker.

enc-X, 2017-01-26:
When I repeat the your test I didn't have such result. I'll add some debug and you will be able to send me more info.

enc-X, 2017-01-26:
Use currnet version (0.06) and send me debug log from the moment when your device reconnect and send IR signal.

bachoo786, 2017-01-27:
hi how do I get the debug log and do you want me to specific IR Signal?

enc-X, 2017-01-27:
Debug log you have available via ESP serial port (the same you use for flashing). In platformio you have serial monitor (menu PlatformIO -> Serial Monitor. Choose proper port and setup speed to 115200.

bachoo786, 2017-01-27:
Here is my log I sent to values “c” and “d”

2A 49 52 3A 20 3D 3D 3D 3D 3D 3D 3D 20 4E 45 57 20 4D 45 53 53 41 47 45 20 3D 3D 3D 3D 3D
 3D 0A 2A 49 52 3A 20 54 6F 70 69 63 3A 20 22 6D 79 64 65 76 69 63 65 73 2F 69 72 74 72 6
1 6E 73 6D 69 74 74 65 72 32 2F 73 65 6E 64 65 72 2F 77 69 70 65 22 0A 2A 49 52 3A 20 4D
65 73 73 61 67 65 3A 20 22 63 22 0A 2A 49 52 3A 20 4C 65 6E 67 74 68 3A 20 31 0A 2A 49 52
 3A 20 50 72 65 66 69 78 3A 20 6D 79 64 65 76 69 63 65 73 2F 69 72 74 72 61 6E 73 6D 69 7
4 74 65 72 32 0A 2A 49 52 3A 20 45 78 74 72 61 63 74 65 64 20 73 75 66 66 69 78 3A 22 2F
73 65 6E 64 65 72 2F 77 69 70 65 22 0A 2A 49 52 3A 20 50 61 79 6C 6F 61 64 20 53 74 72 69
 6E 67 3A 20 22 63 22 0A 2A 49 52 3A 20 57 69 70 65 20 63 6F 6E 66 69 67 0A 2A 49 52 3A 2
0 3D 3D 3D 3D 3D 3D 3D 20 4E 45 57 20 4D 45 53 53 41 47 45 20 3D 3D 3D 3D 3D 3D 0A 2A 49
52 3A 20 54 6F 70 69 63 3A 20 22 6D 79 64 65 76 69 63 65 73 2F 69 72 74 72 61 6E 73 6D 69
 74 74 65 72 32 2F 73 65 6E 64 65 72 2F 77 69 70 65 22 0A 2A 49 52 3A 20 4D 65 73 73 61 6
7 65 3A 20 22 64 22 0A 2A 49 52 3A 20 4C 65 6E 67 74 68 3A 20 31 0A 2A 49 52 3A 20 50 72
65 66 69 78 3A 20 6D 79 64 65 76 69 63 65 73 2F 69 72 74 72 61 6E 73 6D 69 74 74 65 72 32
 0A 2A 49 52 3A 20 45 78 74 72 61 63 74 65 64 20 73 75 66 66 69 78 3A 22 2F 73 65 6E 64 6
5 72 2F 77 69 70 65 22 0A 2A 49 52 3A 20 50 61 79 6C 6F 61 64 20 53 74 72 69 6E 67 3A 20
22 64 22 0A 2A 49 52 3A 20 57 69 70 65 20 63 6F 6E 66 69 67 0A

enc-X, 2017-01-28:
Don't you have message like: 

> *IR: Send GC
> *IR: Elements to send:
> *IR: GC send done.

bachoo786, 2017-01-28:
No I dont I have stuff like this:

\\\\\\"\\\\\\\\\\\\\\\"\\\\\\\"\\\"\""
declare -x declare -x declare -x declare -x declare -x declare -x declare -x declare -x d
eclare -x declare -x __CF_USER_TEXT_ENCODING="\"\\\"\\\\\\\"\\\\\\\\\\\\\\\"\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\"\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\"\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\"\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\"\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\"

0x1F5:0x0:0x2\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\"\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\"\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\
\\\\\\\\\\\\\\\\\\"\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\"\\\\\\
\\\\\\\\\\\\\\\\\\\\\\\\\"\\\\\\\\\\\\\\\"\\\\\\\"\\\"\""

enc-X, 2017-01-29:
That's strange - you have to have messages like I mentioned above. Didn't you disable debug mode in code? Are you sure that it is serial console of ESP device ?

bachoo786, 2017-01-29:
I didnt get any messages that you mentioned above. How do I disable debug mode? is it by (by publishing any value to topic: "prefix/sender/wipe”) ?
It is the serial console of the ESP device i.e. Wemos D1

bachoo786, 2017-01-30:
Are you having the same issue as I am? After my client reconnects to the broker it automatically sends an IR signal. I have set up my TV and led lights so when the broker reboots and the client connects to it after a few seconds, the client then sends or transmits an IR signal without me sending an IR signal.

bachoo786, 2017-01-30:
So my TV and led lights switch on automatically without me sending an IR signal.

enc-X, 2017-02-04:
It may be unclear .... I thing about code level debug mode which is enabled in globals.h  if you have line `#define DEBUG X` you should be able to see debug messages over the serial line - maybe it is wrong speed of line ?

bachoo786, 2017-02-07:
Hi @enc-X 
I think the problem is with the circuit and not the code as I tried with another wemos d1 with just an IR LED (transmitter) attached to pin D5 and GND and I rebooted the client many times and wemos d1 connected to it normally and did not send any IR signals after the client was booted. Do you think the circuit could be a problem?

Just to let you know I am facing this problem with an IR based LED strip lights and my TV.

bachoo786, 2017-02-09:
??