piotrC4/mqtt-ir-transceiver #9: Just getting errors during build

dersch81, 2017-07-14:
Hi, i just get errors by building it, what is the issue with it?

BR Dirk
`
*** [.pioenvs\esp01_1m\src\globals.o] Error 1
*** [.pioenvs\esp01_1m\src\debug.o] Error 1
In file included from src\common.cpp:1:0:
src\globals.h:99:9: error: 'IRrecv' does not name a type

extern IRrecv irrecv;
^
src\globals.h:100:9: error: 'IRsend' does not name a type
extern IRsend irsend;
^
src\globals.h:114:22: error: variable or field 'getIrEncoding' declared void
void  getIrEncoding (decode_results *results, char * result_encoding);
^
src\globals.h:114:22: error: 'decode_results' was not declared in this scope
src\globals.h:114:38: error: 'results' was not declared in this scope
void  getIrEncoding (decode_results *results, char * result_encoding);
^
src\globals.h:114:47: error: expected primary-expression before 'char'
void  getIrEncoding (decode_results *results, char * result_encoding);
^
src\common.cpp:139:22: error: variable or field 'getIrEncoding' declared void
void  getIrEncoding (decode_results *results, char * result_encoding)
^
src\common.cpp:139:22: error: 'decode_results' was not declared in this scope
src\common.cpp:139:38: error: 'results' was not declared in this scope
void  getIrEncoding (decode_results *results, char * result_encoding)
^
src\common.cpp:139:47: error: expected primary-expression before 'char'
void  getIrEncoding (decode_results *results, char * result_encoding)
^
*** [.pioenvs\esp01_1m\src\main.o] Error 1
*** [.pioenvs\esp01_1m\src\common.o] Error 1
 [ERROR] Took 16.66 seconds`

zissouz, 2017-07-15:
I am having similar issues. Tried Atom with PlatformIO and VSCode with PlatformIO. 

I just tested a point fix described here....
https://community.openhab.org/t/mqtt-ir-transmiter-receiver/9433/30

It seems to work....it compiled and uploaded correctly.

enc-X, 2017-07-15:
Latest commit fix the problem, I'll start migration to new version of IRremote library.

closed 2017-07-15 by enc-X

zissouz, 2017-07-15:
Now that is rapid. Thanks enc-X.