piotrC4/mqtt-ir-transceiver #11: Issue with password protected MQTT

dersch81, 2017-08-13:
Hi, is it possible that there is an issue with pw protected MQTT Broker? Since i protected it the IR transreceiver doesn't subscribe anymore.  Of course i added the PW to it. No SSL security. 

The wifi signal is good and the device is connected. So my feeling is there is an issue with pw protected mqtt broker (it is the only device with that behavior)

enc-X, 2017-08-14:
Only one thing which comes to my mind is too long or too complicated password - try simple password.

dersch81, 2017-08-14:
No i don't think so. It is an 9 digit PW with an @ inside. Not sure if the @ make some trouble but all other devices on my MQTT broker does not have any issue with that PW.

closed 2026-10-11 by dersch81