piotrC4/mqtt-ir-transceiver #17: IRSENDRAW

tcpipchip, 2018-08-03:
Sir,

Do this software allow to capture infrared data from no standard hardware and save a RAW code and send the raw code to MQTT server ?

My RAW code got is about 240 words, for example, using IRsendRAW()

TCPIPCHIP

lspaula, 2018-08-30:
Hi. Turns out that if you turn on the raw mode it is possible to read and store code up to 100 words, then it resumes and captures another sequence. I am trying to find out where to change it to make able to store larger codes. Let us hope @enc-X give us a shortcut for that.

Regards,

Luciano

tcpipchip, 2018-08-30:
Thank you so much! Other problem maybe is on the irrecv h lib!

lspaula, 2018-08-30:
While checking the code, I realize this RAWBUF variable which is set to 100 in the IrremoteESP8266.h. Will try changing it to a larger value later today and see if it works.

tcpipchip, 2018-08-30:
Yes, thats a problem too! Thank you so much! Try with ac controllers that return between 240 to 278 ints code

Enviada do meu iPhone

tcpipchip, 2018-08-31:
> I was mistaken in the last message. The file is IRrecv.h
>
> Change
>
> #define RAWBUF 100U // Default length of raw capture buffer
>
> by
>
> #define RAWBUF 512U // Default length of raw capture buffer
>
> and BINGO! Problem solved!
>
> Best regards,
>
> Luciano

Hummm both has that Constant! Well, i test it soon! Thanks to help us!

Enviada do meu iPhone

tcpipchip, 2018-08-31:
Something is wrong. 
What really means these constants ? Need SLOT_SIZE be change to 512U too ?

#define SLOTS_NUMBER 20 // Number of slots
#define SLOT_SIZE 300U   // Size of single slot **<=========**
#define SEQ_SIZE 10     // Raw sequnece size

And, there is a AIR IR CONTROLLER is that always is returning  to **results.decode_type** variable the **15** value .

And to the **results.bits** variable the **24** value ?

What 15 and 24 values means?

lspaula, 2018-08-31:
Sorry pal. I did not change those for mine to work. Slot size of 300 is for saving data to SPIFFS I suppose. It is ok for me. My air conditioner code reaches 217 words only.

Best, 

Luciano

tcpipchip, 2018-09-01:
still have bugs! Because i tested my AIR IR CONTROLLER RAW in another INFRARED C routine and works very nice! On mqtt-ir-transciever not works!

lspaula, 2018-09-01:
Just a dumb question: did you enable raw Mode?

tcpipchip, 2018-09-01:
Good morning

Yes, i did!

I tested many, about 40, but there is an AC IR CONTROLLER that does not work!

CARRIER TYPE RG52B/BGCE

Miguel

tcpipchip, 2018-09-04:
I mean, i can read the RG52B/BGCE, but the if rules to detect if NEC/SONY etc, fails

But RAW is enough to me!