piotrC4/mqtt-ir-transceiver #22: So far exception with any feature

gyurobenjamin, 2020-08-03:
1. Getting the following while trying the receiver:

```
Exception (28):
epc1=0x4022af96 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000e10 depc=0x00000000

LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
  epc1=0x4022af96 in WiFiUDP::endPacket() at /Users/x/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/include/UdpContext.h:441
     (inlined by) WiFiUDP::endPacket() at /Users/x/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiUdp.cpp:176
```

```
0x401005bd in interrupt_handler at /Users/x/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_wiring_digital.cpp:134
0x4022d758 in std::_Sp_counted_deleter<unsigned char*, std::default_delete<unsigned char []>, std::allocator<int>, (__gnu_cxx::_Lock_policy)0>::_M_get_deleter(std::type_info const&) at /Users/x/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr_base.h:367
0x402294b5 in spiffs_page_consistency_check_i at /Users/x/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_check.cpp:746
0x40277959 in txiq_cover at ??:?
0x4022dc53 in countBits(unsigned char const*, unsigned short, bool, unsigned short) at .pio/libdeps/nodemcu/IRremoteESP8266/src/IRutils.cpp:411
0x4020dca6 in esp8266webserver::ESP8266WebServerTemplate<WiFiServer>::_collectHeader(char const*, char const*) at /Users/x/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WebServer/src/Parsing-impl.h:270
0x40214afd in IRrecv::decodeCOOLIX(decode_results*, unsigned short, unsigned short, bool) at .pio/libdeps/nodemcu/IRremoteESP8266/src/ir_Coolix.cpp:671
0x40229bbc in spiffs_gc_find_candidate at /Users/x/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_gc.cpp:278 (discriminator 1)
0x40214c8e in IRrecv::decodeCOOLIX(decode_results*, unsigned short, unsigned short, bool) at .pio/libdeps/nodemcu/IRremoteESP8266/src/ir_Coolix.cpp:700 (discriminator 4)
0x4020dfeb in esp8266webserver::ESP8266WebServerTemplate<WiFiServer>::responseCodeToString(int) at /Users/x/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h:733
0x402018f4 in ArduinoJson::JsonArray::invalid() at .pio/libdeps/nodemcu/ArduinoJson/src/ArduinoJson/JsonArray.hpp:141
0x401002d0 in read_timeout(void*) at .pio/libdeps/nodemcu/IRremoteESP8266/src/IRrecv.cpp:87
0x40101934 in ppEnqueueRxq at ??:?
0x4010534d in wdt_feed at ??:?
0x401002d0 in read_timeout(void*) at .pio/libdeps/nodemcu/IRremoteESP8266/src/IRrecv.cpp:87
0x401053fa in wdt_feed at ??:?
0x401055d1 in ets_timer_arm_new at ??:?
0x401002d0 in read_timeout(void*) at .pio/libdeps/nodemcu/IRremoteESP8266/src/IRrecv.cpp:87
0x40100204 in gpio_intr() at .pio/libdeps/nodemcu/IRremoteESP8266/src/IRrecv.cpp:110
0x40100156 in std::function<void (int)>::operator()(int) const at /Users/x/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/functional:2465
0x401005bd in interrupt_handler at /Users/x/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_wiring_digital.cpp:134
0x401005ac in digitalRead at ??:?
0x40102b6e in wDev_ProcessFiq at ??:?
0x401055d1 in ets_timer_arm_new at ??:?
0x401004e4 in delayMicroseconds at ??:?
0x4021c624 in ESP8266HTTPUpdate::handleUpdate(HTTPClient&, String const&, bool) at /Users/x/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.cpp:275
0x4021a09c in IRrecv::getRClevel(decode_results*, unsigned short*, unsigned short*, unsigned short, unsigned char, short, unsigned short, unsigned char) at .pio/libdeps/nodemcu/IRremoteESP8266/src/ir_RC5_RC6.cpp:259
0x4021a0d5 in IRrecv::getRClevel(decode_results*, unsigned short*, unsigned short*, unsigned short, unsigned char, short, unsigned short, unsigned char) at .pio/libdeps/nodemcu/IRremoteESP8266/src/ir_RC5_RC6.cpp:272
0x401002d0 in read_timeout(void*) at .pio/libdeps/nodemcu/IRremoteESP8266/src/IRrecv.cpp:87
0x4021a0cc in IRrecv::getRClevel(decode_results*, unsigned short*, unsigned short*, unsigned short, unsigned char, short, unsigned short, unsigned char) at .pio/libdeps/nodemcu/IRremoteESP8266/src/ir_RC5_RC6.cpp:272
0x4010104d in realloc at /Users/x/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:823
```

4. With the sending diodes the device just crashed without any message immediately I publish something to the MQTT.

3. Also can't set the device to config mode again.

I'm using nodemcu esp8266.

gyurobenjamin, 2020-08-03:
My dependency graph: 
```
|-- <PubSubClient> 2.8
|-- <WifiManager> 0.15.0
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ArduinoJson> 5.13.4
|-- <IRremoteESP8266> 2.7.9 #857a9ee
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
|-- <EEPROM> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266httpUpdate> 1.3
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
```

octavius08, 2021-01-23:
Hi,
Did you setup the files correctly before compiling for D1_mini, please check the Issue #24.
Maybe the GPIOs are set incorrectly for the nodemcu? Otherwise try a Wemos D1 mini.